csatt/IV_Swinger #177: Use pvlib for I-V calculations?

kanderso-nrel, 2021-07-29:
Hi there, I just came across this project (very cool by the way!) and wanted to float the idea of using `pvlib` for the I-V curve modeling part of this project.  I see a few benefits:
- You could use other single-diode models besides the De Soto model (which I think is what IV_Swinger uses? I only very quickly skimmed the code) like the CEC and PVSyst models: https://pvlib-python.readthedocs.io/en/stable/api.html#single-diode-models
- pvlib ships with a copy of the CEC module database (SDM parameters for 20,000+ modules).  Here's an example of using pvlib's CEC modeling code to generate I-V curves: https://pvsc-python-tutorials.github.io/PVSC48-Python-Tutorial/Tutorial%204%20-%20Model%20a%20Module%27s%20Performance.html#cec-model-aka-sam-or-6-parameter-model
- Perhaps pvlib's code is better-vetted -- I only say this because I happened to notice #167 and #168.
- Not sure this is relevant, but maybe: access to other modeling functionality aside from I-V curves, like irradiance/spectrum/temperature and such.  Maybe clear-sky irradiance modeling would be of interest?

However, pvlib has its own dependencies, so it would add some bloat to this project's overall dependency list.  Here's pvlib's set of basic dependencies: https://github.com/pvlib/pvlib-python/blob/master/setup.py#L41-L45

Anyway happy to discuss more if you're interested, or feel free to close the issue if you're not :)

csatt, 2021-07-30:
Hi Kevin,

I discovered pvlib just as I was completing the PV modeling feature. I really don't know how I had not stumbled on it earlier. My first thought was that I had wasted many hours doing it myself, but then I remembered how much I had learned in the process.

I agree with all of your points, especially the one on vetting. I will not close this Issue, but I may or may not get around to working on it. For now, I think what I have serves its purpose. Actually, I am not even sure if anyone is using the feature yet. I really don't have a very large user base.

I'm glad you reached out, and it's great to meet someone at NREL. On the one hand, I think of IV Swinger as a toy. But it really has been useful for some people, especially in other countries where buying a commercial IV curve tracer is out of the question. I do think it could be useful to more people, but I haven't put much effort into publicizing it. BTW, how did you find it?

  - Chris

kanderso-nrel, 2021-07-30:
>how did you find it?

I was browsing GitHub's search results for repositories that mention "photovoltaic" and IV Swinger caught my eye.  I'd been playing with the idea of making a barebones curve tracer using a microcontroller or single-board computer and was happy to see someone had already done the hard work for me :)  Not sure when (or if) I'll have time to actually sit down and build one, but I hope to before too long.  

I know the original motivation for this project was related to a class at Stanford; do you have a vision for where you want it to go in the future?  I'm curious about both the technical (software/hardware evolution) and non-technical (intended audience, outreach etc) aspects of that question.

csatt, 2021-07-31:
That would be great if you built an IV Swinger 2! The only catch is that due to the global semiconductor shortage, the SSR parts are not currently available anywhere. Maybe by the time you are ready, they will be. It should still be possible to build the EMR version.

I never had a clear vision for this project. I originally (2014) thought it would be a fun thing to do for a couple of months. Like a lot of things, it ended up taking on a life of its own, and I followed where it went more than taking it in a particular direction. When I put it in the public domain, I thought it would be used for undergrad classes like the one at Stanford. I had my USA blinders on and didn't expect the vast majority of interest to be from other countries for graduate research. That has been very cool.

Moving forward, I have mixed feelings. I'm 60 years old. I retired early, and this has been a great way to stay technical and also feel like I'm doing something that has at least some value. But I have some ideas for other projects, and I may want to put this one on the shelf. On the other hand, after having put so much time into it, I would like more people to be able to use it. One barrier is that not everyone is capable of building the hardware. I don't want to spend my time building them for sale and I don't have a desire to make a business out of it. I've always thought it would be great if someone else wanted to build them and sell them for whatever they could charge. I might spend some cycles trying to find someone like that. I also think it would be nice to have a website, so I might do that.

As for the hardware and software evolution, there's a bit of history at the beginning of the document, "IV Swinger 2: Design and Theory of Operation" (https://github.com/csatt/IV_Swinger/blob/master/docs/IV_Swinger2/IV_Swinger2_Design.pdf)